FabLabCasaCorsini/bracciorobotmatera #38: Modifica files GCode

CoffeePerry, 2020-02-13:
### Modifica files GCode

A seguito del recente cambio della sorgente luminosa della torcia, è necessario riadattare la generazione dei files GCode ed i files GCode esistenti sul raspberry (offline), per quanto riguarda il comando di Grbl:

`S900`

il quale deve essere sostituito con:

`S1000`

In quanto la nuova sorgente luminosa necessita di un _duty cycle_ del 100% per emettere luce.

albtrentadue, 2020-02-14:
Questa è una modifica da apportare all'algoritmo di conversione implementato dall'App Android

CoffeePerry, 2020-02-15:
Ho controllato nel codice dell'App Android, ed ho notato che invia al server stringhe del tipo:

`0.0-96.682465-0.0-96.445496-0.9569378-92.891-5.5023923-79.85782-13.875598-60.900475-21.291866-43.127964-29.425838-26.06635-36.1244-15.639811-37.79904-13.744076$$55.023922-100.0-55.50239-98.34123-60.765553-85.54502-71.29186-60.900475-83.97129-33.886257-93.30143-12.559241-100.0-0.0`

Ovvero stringhe di sole coordinate dei punti da disegnare (e _$$_ come separatore di linee), pertanto sospetto che la modifica da attuare sia nel file _php/save.php_ sul server oppure in uno dei file _*.py_ sul Raspberry. Ora li controllo.

CoffeePerry, 2020-02-15:
A mio parere, pure il file _php/save.php_ non necessita di modifiche, in quanto, quando il server riceve una richiesta POST dall'App Android su _save.php_, quest'ultimo produce due file (senza _.txt_ alla fine del nome dei file):

**coord**:
[test-_-7a37bcce-5d79-468a-9f30-ba51a48f1433.coord.txt](https://github.com/FabLabCasaCorsini/bracciorobotmatera/files/4208946/test-_-7a37bcce-5d79-468a-9f30-ba51a48f1433.coord.txt)

**gcode**:
[test-_-7a37bcce-5d79-468a-9f30-ba51a48f1433.gcode.txt](https://github.com/FabLabCasaCorsini/bracciorobotmatera/files/4208947/test-_-7a37bcce-5d79-468a-9f30-ba51a48f1433.gcode.txt)

Il file _coord_ contiene la stringa inviata dall'App Android, mentre nel file _gcode_, vedo che non vengono scritti comandi `S` (Spindle Speed).

Procedo quindi a controllare che l'aggiunta dei vari comandi `S` non sia opera di un qualche script Python sul Raspberry.

CoffeePerry, 2020-02-20:
Risolto da @aappeddu il 19/02/2020 sera, il codice su quale stavamo lavorando non era aggiornato.

closed 2020-02-20 by CoffeePerry